alibaba/tidevice #373: macOs docker环境为ubuntu，执行tidevice list抛出tidevice.exceptions.SocketError: socket unix:/var/run/usbmuxd unable to connect异常

csushiye, 2024-03-15:
报错堆栈：
Traceback (most recent call last):
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 60, in __init__
    self._sock = self._connect(addr)
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 78, in _connect
    raise SocketError("socket unix:{} unable to connect".format(addr))
tidevice.exceptions.SocketError: socket unix:/var/run/usbmuxd unable to connect

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/androidusr/.local/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/__main__.py", line 1027, in main
    actions[args.subparser](args)
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/__main__.py", line 84, in cmd_list
    ds = um.device_list()
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 80, in device_list
    data = self.send_recv(payload, timeout=10)
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 52, in send_recv
    s = self.create_connection()
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 48, in create_connection
    psock = PlistSocket(self.__address, self._next_tag())
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 183, in __init__
    super().__init__(addr)
  File "/home/androidusr/.local/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 62, in __init__
    raise SocketError("socket connect error") from e

docker环境是ubuntu20，起docker命令是：sudo docker run -it --name env_ios_02 -d -v /var:var/ -e TZ="Asia/Shanghai" --network host --privileged appium/appium，把宿主机的/var/run/usbmuxd套接字文件挂载到docker中，这样docker执行tidevice就可以想ubsmux服务发送请求呢，但还是会报这个错误，有在macOs机器 docker linux环境中通过tidevice进行自动化测试吗

codeskyblue, 2024-03-19:
别试了，mac上docker不能共享unix socket

csushiye, 2024-03-22:
> 别试了，mac上docker不能共享unix socket

已经放弃了，改成还是在mac宿主机上识别设备了

closed 2024-04-04 by codeskyblue

lixf6, 2024-11-07:
提供一种实现方案：（已验证）
宿主机：TCP-LISTEN:12345,reuseaddr,fork UNIX-CONNECT:/var/run/usbmuxd
切记不要映射/var/run/usbmuxd
容器内：socat UNIX-LISTEN:/var/run/usbmuxd,fork TCP:host.docker.internal:12345
通过这个绕过去的方式，是可以读取到设备的，已实测；