alibaba/tidevice #374: tidevice.exceptions.MuxServiceError: InvalidService

ZeeLyn, 2024-03-23:
```shell
[W 240323 12:00:44 _wdaproxy:65] [00008101-001A350A36D0001E] Unknown exception: request() got an unexpected keyword argument 'chunked'
[E 240323 12:00:44 _wdaproxy:153] [00008101-001A350A36D0001E] wda started failed
[D 240323 12:00:44 _wdaproxy:134] [00008101-001A350A36D0001E] launch WDA
[W 240323 12:00:44 _wdaproxy:65] [00008101-001A350A36D0001E] Unknown exception: request() got an unexpected keyword argument 'chunked'
[I 240323 12:00:45 _device:996] BundleID: com.zeelyn.WebDriverAgentRunner.xctrunner
[I 240323 12:00:45 _device:1023] ProductVersion: 17.4.1
[I 240323 12:00:45 _device:1024] UDID: 00008101-001A350A36D0001E
Traceback (most recent call last):
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\_device.py", line 501, in start_service
    return self._unsafe_start_service(name)
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\_device.py", line 520, in _unsafe_start_service
    raise MuxServiceError(error)
tidevice.exceptions.MuxServiceError: InvalidService

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "F:\Program Files\Python\Python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "F:\Program Files\Python\Python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\__main__.py", line 1032, in <module>
    main()
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\__main__.py", line 1027, in main
    actions[args.subparser](args)
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\__main__.py", line 300, in cmd_xcuitest
    d.runwda(args.bundle_id,
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\_device.py", line 997, in runwda
    return self.xcuitest(bundle_id, target_bundle_id=target_bundle_id,
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\_device.py", line 1034, in xcuitest
    x1 = self._connect_testmanagerd_lockdown()
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\_device.py", line 802, in _connect_testmanagerd_lockdown
    conn = self.start_service(
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\_device.py", line 503, in start_service
    self.mount_developer_image()
  File "E:\GitHub\AndroidAppReptile\.venv\lib\site-packages\tidevice\_device.py", line 674, in mount_developer_image
    raise ServiceError("DeveloperImage not found")
tidevice.exceptions.ServiceError: DeveloperImage not found
[W 240323 12:00:45 _wdaproxy:65] [00008101-001A350A36D0001E] Unknown exception: request() got an unexpected keyword argument 'chunked'
[E 240323 12:00:45 _wdaproxy:153] [00008101-001A350A36D0001E] wda started failed

```

codeskyblue, 2024-04-08:
iOS17后面不支持了，需要更换tidevice3

ghb0224, 2024-04-24:
> iOS17后面不支持了，需要更换tidevice3

安装了tidevice3后， 该指令tidevice wdaproxy -B  com.facebook.WebDriverAgentRunner.iphone703.xctrunner --port 8100还是报错，wda start failed,
IOS 17.4
WIN10环境下

zzdzzd123, 2025-07-18:
解决了吗？我也是这个问题