alibaba/tidevice #377: `tidevice`不兼容高版本的`pyOpenSSL`，大部分命令都会报：Invalid version. The only valid version for X509Req is 0

vankillua, 2024-04-03:
Mac M1安装`tidevice`，命令：`pip install -U "tidevice[openssl]"`，结果如下：
```
Collecting tidevice[openssl]
  Using cached tidevice-0.12.8-py3-none-any.whl (94 kB)
Collecting simple-tornado>=0.2.2
  Using cached simple_tornado-0.2.2-py3-none-any.whl (3.3 kB)
Collecting tornado
  Downloading tornado-6.4-cp38-abi3-macosx_10_9_universal2.whl (433 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 433.1/433.1 KB 11.7 MB/s eta 0:00:00
Collecting packaging
  Using cached packaging-24.0-py3-none-any.whl (53 kB)
Collecting retry
  Using cached retry-0.9.2-py2.py3-none-any.whl (8.0 kB)
Collecting tabulate
  Using cached tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting Pillow
  Downloading pillow-10.3.0-cp39-cp39-macosx_11_0_arm64.whl (3.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.4/3.4 MB 2.6 MB/s eta 0:00:00
Collecting logzero
  Using cached logzero-1.7.0-py2.py3-none-any.whl (16 kB)
Collecting requests
  Using cached requests-2.31.0-py3-none-any.whl (62 kB)
Collecting colored
  Downloading colored-2.2.4-py3-none-any.whl (16 kB)
Collecting deprecation
  Using cached deprecation-2.1.0-py2.py3-none-any.whl (11 kB)
Collecting simplejson
  Downloading simplejson-3.19.2-cp39-cp39-macosx_11_0_arm64.whl (75 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 75.5/75.5 KB 9.6 MB/s eta 0:00:00
Collecting pyasn1
  Downloading pyasn1-0.6.0-py2.py3-none-any.whl (85 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 85.3/85.3 KB 13.0 MB/s eta 0:00:00
Collecting pyOpenSSL
  Downloading pyOpenSSL-24.1.0-py3-none-any.whl (56 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 56.9/56.9 KB 7.4 MB/s eta 0:00:00
Collecting cryptography<43,>=41.0.5
  Downloading cryptography-42.0.5-cp39-abi3-macosx_10_12_universal2.whl (5.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.9/5.9 MB 2.7 MB/s eta 0:00:00
Collecting urllib3<3,>=1.21.1
  Downloading urllib3-2.2.1-py3-none-any.whl (121 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 121.1/121.1 KB 18.3 MB/s eta 0:00:00
Collecting idna<4,>=2.5
  Using cached idna-3.6-py3-none-any.whl (61 kB)
Collecting charset-normalizer<4,>=2
  Downloading charset_normalizer-3.3.2-cp39-cp39-macosx_11_0_arm64.whl (120 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 120.4/120.4 KB 12.9 MB/s eta 0:00:00
Collecting certifi>=2017.4.17
  Using cached certifi-2024.2.2-py3-none-any.whl (163 kB)
Collecting py<2.0.0,>=1.4.26
  Using cached py-1.11.0-py2.py3-none-any.whl (98 kB)
Collecting decorator>=3.4.2
  Using cached decorator-5.1.1-py3-none-any.whl (9.1 kB)
Collecting cffi>=1.12
  Downloading cffi-1.16.0-cp39-cp39-macosx_11_0_arm64.whl (176 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 176.8/176.8 KB 32.9 MB/s eta 0:00:00
Collecting pycparser
  Downloading pycparser-2.22-py3-none-any.whl (117 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 117.6/117.6 KB 15.0 MB/s eta 0:00:00
Installing collected packages: logzero, urllib3, tornado, tabulate, simplejson, pycparser, pyasn1, py, Pillow, packaging, idna, decorator, colored, charset-normalizer, certifi, simple-tornado, retry, requests, deprecation, cffi, tidevice, cryptography, pyOpenSSL
Successfully installed Pillow-10.3.0 certifi-2024.2.2 cffi-1.16.0 charset-normalizer-3.3.2 colored-2.2.4 cryptography-42.0.5 decorator-5.1.1 deprecation-2.1.0 idna-3.6 logzero-1.7.0 packaging-24.0 py-1.11.0 pyOpenSSL-24.1.0 pyasn1-0.6.0 pycparser-2.22 requests-2.31.0 retry-0.9.2 simple-tornado-0.2.2 simplejson-3.19.2 tabulate-0.9.0 tidevice-0.12.8 tornado-6.4 urllib3-2.2.1
WARNING: You are using pip version 22.0.4; however, version 24.0 is available.
You should consider upgrading via the '/Users/killua/.pyenv/versions/3.9.16/bin/python3.9 -m pip install --upgrade pip' command.
```

执行`tidevice list`报错，如下：
```
Traceback (most recent call last):
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_device.py", line 246, in handshake
    self._pair_record = self._read_pair_record()
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_device.py", line 173, in _read_pair_record
    data = self._usbmux.send_recv(payload)
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_usbmux.py", line 54, in send_recv
    self._check(data)
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_usbmux.py", line 98, in _check
    raise MuxReplyError(data['Number'])
tidevice.exceptions.MuxReplyError: UsbmuxReplyCode.BadDevice

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/killua/.pyenv/versions/3.9.16/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/__main__.py", line 1027, in main
    actions[args.subparser](args)
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/__main__.py", line 101, in cmd_list
    serial = _d.get_value("SerialNumber")
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_device.py", line 366, in get_value
    with self.create_session() as conn:
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_device.py", line 312, in create_session
    "HostID": self.pair_record['HostID'],
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_device.py", line 137, in pair_record
    self.handshake()
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_device.py", line 249, in handshake
    self._pair_record = self.pair()
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_device.py", line 203, in pair
    cert_pem, priv_key_pem, dev_cert_pem = make_certs_and_key(device_public_key)
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_ca.py", line 32, in make_certs_and_key
    host_req = make_req(root_key)
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tidevice/_ca.py", line 83, in make_req
    req.set_version(2)
  File "/Users/killua/.pyenv/versions/3.9.16/lib/python3.9/site-packages/OpenSSL/crypto.py", line 1038, in set_version
    raise ValueError(
ValueError: Invalid version. The only valid version for X509Req is 0.
```

具体原因：
[image]

[image]

[image]

解决方案：
1. 指定`pyOpenSSL`版本为`23.1.1`
2. 修改`tidevice`代码，`req.set_version(2)`改为`req.set_version(0)`

codeskyblue, 2024-04-08:
要不来个pr

codeskyblue, 2024-04-12:
published as version 0.12.9

codeskyblue, 2024-04-12:
thanks @vankillua

closed 2024-04-12 by codeskyblue

991547436, 2024-04-18:
pip install urllib3==1.21.1

hhstore, 2024-06-16:
- M1 Mac 在, 最新版本 `tidevice>=0.12.10`, 执行 `tidevice pair` 依然报这个错误. 

```
ValueError: Invalid version. The only valid version for X509Req is 0.

```

commit 465869d9 (2024-04-12): req.set_version(0), close #377