alibaba/tidevice #390: 启动wda连接失败，有没有大佬看下原因

PatrickHXH, 2024-12-04:
使用命令tidevice xctest -B com.autoui.test.autoui.WebDriverAgentRunner1.xctrunner

[I 241204 09:47:08 _device:996] BundleID: com.autoui.test.autoui.WebDriverAgentRunner1.xctrunner
[I 241204 09:47:08 _device:1023] ProductVersion: 16.0.3
[I 241204 09:47:08 _device:1024] UDID: 00008101-001D2CCC0ED8001E
[I 241204 09:47:08 _imagemounter:25] Download https://github.com/JinjunHan/iOSDeviceSupport/raw/master/iOSDeviceSupport/16.0.zip -> C:\Users\86492\.tidevice\device-support\16.0.zip
Traceback (most recent call last):
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 501, in start_service
    return self._unsafe_start_service(name)
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 520, in _unsafe_start_service
    raise MuxServiceError(error)
tidevice.exceptions.MuxServiceError: InvalidService

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Program Files (x86)\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Program Files (x86)\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\Program Files (x86)\Python39\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\__main__.py", line 1027, in main
    actions[args.subparser](args)
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\__main__.py", line 300, in cmd_xcuitest
    d.runwda(args.bundle_id,
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 997, in runwda
    return self.xcuitest(bundle_id, target_bundle_id=target_bundle_id,
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 1034, in xcuitest
    x1 = self._connect_testmanagerd_lockdown()
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 802, in _connect_testmanagerd_lockdown
    conn = self.start_service(
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 503, in start_service
    self.mount_developer_image()
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 661, in mount_developer_image
    developer_img_dir = self._request_developer_image_dir(int(major), guess_minor)
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_device.py", line 625, in _request_developer_image_dir
    image_path = get_developer_image_path(version)
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_imagemounter.py", line 102, in get_developer_image_path
    image_zip_path = cache_developer_image(version)
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_imagemounter.py", line 91, in cache_developer_image
    raise err
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_imagemounter.py", line 79, in cache_developer_image
    _urlretrieve(url, image_zip_path)
  File "D:\Program Files (x86)\Python39\lib\site-packages\decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "D:\Program Files (x86)\Python39\lib\site-packages\retry\api.py", line 73, in retry_decorator
    return __retry_internal(partial(f, *args, **kwargs), exceptions, tries, delay, max_delay, backoff, jitter,
  File "D:\Program Files (x86)\Python39\lib\site-packages\retry\api.py", line 33, in __retry_internal
    return f()
  File "D:\Program Files (x86)\Python39\lib\site-packages\tidevice\_imagemounter.py", line 30, in _urlretrieve
    with requests.get(url, headers=headers, stream=True, timeout=_REQUESTS_TIMEOUT) as r:
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\sessions.py", line 723, in send
    history = [resp for resp in gen]
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\sessions.py", line 723, in <listcomp>
    history = [resp for resp in gen]
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\sessions.py", line 266, in resolve_redirects
    resp = self.send(
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "D:\Program Files (x86)\Python39\lib\site-packages\requests\adapters.py", line 520, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /JinjunHan/iOSDeviceSupport/master/iOSDeviceSupport/16.0.zip (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x0000020277C3D400>: Failed to establish a new connection: [Errno 11004] getaddrinfo failed'))

Cqjun11, 2024-12-26:
同，有找到解决方案吗

Blue-FatMan, 2025-06-17:
找个代理，用代理连接网络